ibm-genwqe/genwqe-user #16: file information (stat) is not preserved

sebott, 2015-09-25:
# ll foo 
-rwxr-xr-x 1 root root 95349752 Sep 25 13:24 foo
# src/genwqe-user/tools/genwqe_gzip foo
# src/genwqe-user/tools/genwqe_gunzip foo.gz
# ll foo 
-rw-r--r-- 1 root root 95349752 Sep 25 13:24 foo

fhaverkamp, 2015-09-25:
Yes, that looks like a bug.

ibm-genwqe, 2015-09-28:
please set or check if

"git config core.filemode=true"

is true

Mit freundlichen Grüßen / Kind regards

Eberhard S. Amann

zEDC Firmware
IBM Systems &Technology Group, Integrated Systems Development
Opens Systems Hardware Development
                                                                                             
                                                                                             
                                                                                             
                                                                                             
                                                                                             
 Phone:            +49-7034-274-3029          IBM Deutschland                      (Embedded 
                                                                                 image moved 
                                                                                    to file: 
                                                                               pic59535.gif) 
                                                                                             
 E-Mail:           esa@de.ibm.com             Schoenaicher Str. 220                          
                                                                                             
                                              71032 Boeblingen                               
                                                                                             
                                              Germany                                        
                                                                                             
                                                                                             
                                                                                             
                                                                                             
                                                                                             
 IBM Deutschland                                                                             
 Research &                                                                                  
 Development                                                                                 
 GmbH /                                                                                      
 Vorsitzender des                                                                            
 Aufsichtsrats:                                                                              
 Martina Koederitz                                                                           
                                                                                             
 Geschäftsführung:                                                                           
 Dirk Wittkopp                                                                               
 Sitz der                                                                                    
 Gesellschaft:                                                                               
 Böblingen /                                                                                 
 Registergericht:                                                                            
 Amtsgericht                                                                                 
 Stuttgart, HRB                                                                              
 243294                                                                                      

            (#16)

sebott, 2015-09-28:
> please set or check if
> 
> "git config core.filemode=true"
> 
> is true

The genwqe-user files are fine. It's that genwqe_gzip modifies the
file attributes:

# ll foo
-rwxr-xr-x 1 root root 95349752 Sep 25 13:24 foo
# src/genwqe-user/tools/genwqe_gzip foo
# src/genwqe-user/tools/genwqe_gunzip foo.gz
# ll foo
-rw-r--r-- 1 root root 95349752 Sep 25 13:24 foo

regards,
Sebastian

fhaverkamp, 2015-09-28:
Hi Sebastian,

commit 919344155ed44e88e9e160d81c427996d513bbdc contains a fix proposal. Please let me know if that works out for you. I have already requested merge ;-). So we need to stop Eberhard, if that was too quick.

Thanks

Frank

closed 2015-09-28 by fhaverkamp

sebott, 2015-09-28:
Looks good!
Thanks for the quick fix,
Sebastian