ibm-genwqe/genwqe-user #26: samtools: CAPI version not working properly

fhaverkamp, 2015-10-09:
haver@tul2eth3:~/genomics$ time -p ZLIB_INFLATE_IMPL=0x61 ZLIB_DEFLATE_IMPL=0x61 ZLIB_ACCELERATOR=GENWQE LD_PRELOAD=/opt/genwqe/lib/genwqe/libz.so:/opt/genwqe/lib/libcxl.so samtools sort -@41 -T NA1.bam.tmp -O bam -o NA1.sorted.bam NA1.bam
real 10.97
user 16.43
sys 22.90
haver@tul2eth3:~/genomics$ time -p ZLIB_INFLATE_IMPL=0x61 ZLIB_DEFLATE_IMPL=0x61 ZLIB_ACCELERATOR=SW LD_PRELOAD=/opt/genwqe/lib/genwqe/libz.so:/opt/genwqe/lib/libcxl.so samtools sort -@41 -T NA1.bam.tmp -O bam -o NA1.sorted.bam NA1.bam
real 10.86
user 12.15
sys 19.11
haver@tul2eth3:~/genomics$ time -p ZLIB_INFLATE_IMPL=0x61 ZLIB_DEFLATE_IMPL=0x61 ZLIB_ACCELERATOR=CAPI LD_PRELOAD=/opt/genwqe/lib/genwqe/libz.so:/opt/genwqe/lib/libcxl.so samtools sort -@41 -T NA1.bam.tmp -O bam -o NA1.sorted.bam NA1.bam
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
00018764.0001878a deflate.c:487: Error: outp_returned (0) invalid
hardware.c:651: Error: not all input absorbed!
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[... 140 lines cut ...]
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
00018764.00018766 deflate.c:487: Error: outp_returned (0) invalid
hardware.c:651: Error: not all input absorbed!
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
[zedc_deflate] What a pitty optimization did not work
00018764.00018764 deflate.c:754: Error: deflate failed rc=-402
  DDCB returned (RETC=000 ATTN=0002 PROGR=1) ERR
hardware.c:651: Error: not all input absorbed!
00018764.00018780 deflate.c:754: Error: deflate failed rc=-402
  DDCB returned (RETC=000 ATTN=0002 PROGR=1) ERR
hardware.c:651: Error: not all input absorbed!
00018764.0001877f deflate.c:754: Error: deflate failed rc=-402
  DDCB returned (RETC=000 ATTN=0002 PROGR=1) ERR
hardware.c:651: Error: not all input absorbed!
^C
real 25.67
user 6.26
sys 2.40

Something seems not working right in our CAPI version. We need to investigate and fix this.

ibm-genwqe, 2015-10-21:
I am providing a patch to clean this issue

fhaverkamp, 2015-10-21:
In fact the samtools failure here was exposing multiple problems. In our original version we had reference counter issues leading to double free situation when releasing resources. After this was fixed the issue Eberhard mentioned above remained. This is caused by a deficiency in the cxl Linux device driver. I tried to describe this in https://github.com/ibm-capi/libcxl/issues/7. Eberhards latest fixes are indeed a circumvention for this restriction.

closed 2015-10-21 by fhaverkamp