ibm-genwqe/genwqe-user #32: deflate Z_FULL_FLUSH performance not as good as possible

fhaverkamp, 2015-11-05:
We measured that the possible deflate bandwidth was not as expected when we used Z_FULL_FLUSH. We created the following patch to fix this:
https://github.com/ibm-genwqe/genwqe-user/commit/7b32d996ec35aadcfcf0154cb42f6f5e885ec319

    ZLIB_FLAG_OMIT_LAST_DICT = 0x40

At this point in time we require that the ZLIB_FLAG_OMIT_LAST_DICT in the ZLIB_DEFLATE_IMPL environment variable must be set to activate this feature. So use this with:

    export ZLIB_DEFLATE_IMPL=0x41

thilomaurer, 2015-11-06:
Pleasure

fhaverkamp, 2015-11-12:
I have merged it into master for testing and tryout. For me and Thilo it already worked nicely. We do not want to run our test without it ;-).

closed 2015-11-12 by fhaverkamp