ibm-genwqe/genwqe-user #34: Parallel Make failes

thilomaurer, 2015-11-06:
This is usually the case when there are unspecified but necessary dependencies between targets in the Makefile.
```
root@tul2eth3:~/genwqe-user# make -j
make[1]: Entering directory '/root/genwqe-user/lib'
make[1]: Entering directory '/root/genwqe-user/tools'
	[CFG]	zlib-1.2.8
make[2]: Entering directory '/root/genwqe-user/lib'
make[2]: Entering directory '/root/genwqe-user/lib'
make[2]: Entering directory '/root/genwqe-user/lib'
	[CC]	genwqe_gunzip.o
	[INFO]	Using ../../libcxl/libcxl.a.
	[CC]	libzHW.o
	[CC]	deflate.o
	[CC]	inflate.o
	[CC]	libcard.o
	[CC]	software.o
	[CC]	libddcb.o
	[CC]	ddcb_card.o
	[CC]	hardware.o
	[CC]	wrapper.o
	[CC]	ddcb_capi.o
	[INFO]	Using ../../libcxl/libcxl.a.
	[CC]	libzHW.o
	[CC]	inflate.o
	[CC]	deflate.o
	[INFO]	Using ../../libcxl/libcxl.a.
	[CC]	libcard.o
	[CC]	libddcb.o
	[CC]	libzHW.o
	[CC]	ddcb_card.o
	[INFO]	Using ../../libcxl/libcxl.a.
	[CC]	ddcb_capi.o
	[CC]	inflate.o
	[CC]	libzHW.o
	[CC]	inflate.o
	[CC]	wrapper.o
	[CC]	deflate.o
	[CC]	deflate.o
	[CC]	libcard.o
	[CC]	hardware.o
	[CC]	libcard.o
	[CC]	software.o
	[CC]	libddcb.o
	[CC]	ddcb_card.o
	[CC]	libddcb.o
	[CC]	ddcb_capi.o
	[CC]	ddcb_card.o
	[CC]	wrapper.o
	[CC]	wrapper.o
	[CC]	ddcb_capi.o
	[CC]	software.o
	[CC]	hardware.o
	[CC]	hardware.o
	[CC]	software.o
	[CC]	libddcb.o
	[CC]	libddcb.o
	[CC]	ddcb_card.o
	[CC]	ddcb_card.o
	[CC]	ddcb_card.o
	[CC]	ddcb_card.o
	[CC]	ddcb_capi.o
	[CC]	inflate.o
	[CC]	hardware.o
	[CC]	hardware.o
	[LD]	__libDDCB.o
	[CC]	libDDCB-4.0.6-3-g665b.so
	[CC]	libddcb.o
	[CC]	libddcb.o
	[CC]	libcard.o
	[LD]	__libcard.o
	[CC]	libcard-4.0.6-3-g665b.so
	[CC]	libzHW.o
	[CC]	libzHW.o
	[CC]	libzHW.o
	[CC]	genwqe_gunzip
gcc: error: ../lib/libzADC.a: No such file or directory
gcc: error: ../lib/libzHW.a: No such file or directory
Makefile:60: recipe for target 'genwqe_gunzip' failed
make[1]: *** [genwqe_gunzip] Error 1
make[1]: *** Waiting for unfinished jobs....
	[CC]	libzHW.o
	[CC]	deflate.o
	[CC]	deflate.o
	[CC]	deflate.o
	[CC]	deflate.o
	[LD]	__libzHW.o
	[CC]	libzHW-4.0.6-3-g665b.so
	[CC]	inflate.o
	[LD]	__libzHW.o
	[CC]	libzHW-4.0.6-3-g665b.so
	[CC]	ddcb_capi.o
	[LD]	__libDDCB.o
	[CC]	inflate.o
	[CC]	libDDCB-4.0.6-3-g665b.so
	[CC]	inflate.o
	[CC]	ddcb_capi.o
	[CC]	ddcb_capi.o
	[CC]	hardware.o
	[LD]	__libzHW.o
	[LD]	__libzHW.o
	[CC]	libzHW-4.0.6-3-g665b.so
	[CC]	libzHW-4.0.6-3-g665b.so
	[CC]	libcard.o
	[LD]	__libDDCB.o
	[LD]	__libDDCB.o
	[CC]	libDDCB-4.0.6-3-g665b.so
	[CC]	hardware.o
	[CC]	libDDCB-4.0.6-3-g665b.so
	[CC]	wrapper.o
	[LD]	__libcard.o
	[CC]	wrapper.o
	[CC]	libcard-4.0.6-3-g665b.so
	[CC]	wrapper.o
	[CC]	wrapper.o
	[CC]	software.o
	[CC]	software.o
	[LD]	__libzADC.o
software.o: file not recognized: File truncated
Makefile:116: recipe for target '__libzADC.o' failed
make[2]: *** [__libzADC.o] Error 1
make[2]: Leaving directory '/root/genwqe-user/lib'
Makefile:53: recipe for target '../lib/libDDCB.a' failed
make[1]: *** [../lib/libDDCB.a] Error 2
	[CC]	libcard.o
	[CC]	libcard.o
	[CC]	software.o
	[LD]	__libzADC.o
	[LD]	__libcard.o
	[LD]	__libcard.o
	[CC]	libcard-4.0.6-3-g665b.so
	[CC]	libcard-4.0.6-3-g665b.so
	[LD]	__libzADC.o
	[CC]	libzADC-4.0.6-3-g665b.so
__libzADC.o: file not recognized: File format not recognized
	[CC]	libzADC-4.0.6-3-g665b.so
collect2: error: ld returned 1 exit status
Makefile:125: recipe for target 'libzADC-4.0.6-3-g665b.so' failed
make[2]: *** [libzADC-4.0.6-3-g665b.so] Error 1
make[2]: Leaving directory '/root/genwqe-user/lib'
Makefile:53: recipe for target '../lib/libzHW.a' failed
make[1]: *** [../lib/libzHW.a] Error 2
make[2]: Leaving directory '/root/genwqe-user/lib'
make[1]: Leaving directory '/root/genwqe-user/tools'
Makefile:84: recipe for target 'tools' failed
make: *** [tools] Error 1
make: *** Waiting for unfinished jobs....
	[CC]	software.o
	[LD]	__libzADC.o
	[CC]	libzADC-4.0.6-3-g665b.so
make[1]: Leaving directory '/root/genwqe-user/lib'
```

fhaverkamp, 2015-11-12:
I think this is caused somehow by the way we handle linking/building libraries we need. Thilo any good idea how this is to be done right? I am unsure how to fix this. But you are certainly right that it would be nice to exploit the parallelism our huge systems provide during software build.

fhaverkamp, 2015-11-25:
Thilo contributed a fix for this himself. Thanks:
https://github.com/ibm-genwqe/genwqe-user/commit/4c7e6a96eaf970ad403ca5e2d5546f56a4939116

closed 2015-11-25 by fhaverkamp