ibm-genwqe/genwqe-user #60: Cross compiling the code does not work yet

fhaverkamp, 2016-01-29:
Vaibhav tried to cross compile our code. He failed badly:

vajain21@vajain21  ~/source/genwqe-user   master  make CROSS=powerpc64le-unknown-linux-gnu-
make[1]: Entering directory '/home/vajain21/source/genwqe-user/lib'
[CC] libzHW.o
[CC] libzHW.o
[CC] inflate.o
[CC] inflate.o
[CC] deflate.o
[CC] deflate.o
[LD] __libzHW.o
powerpc64le-unknown-linux-gnu-ld: unrecognised emulation mode: elf_x86_64
Supported emulations: elf64lppc elf32lppc elf32lppclinux elf32lppcsim
Makefile:77: recipe for target '__libzHW.o' failed
make[1]: *** [__libzHW.o] Error 1
make[1]: Leaving directory '/home/vajain21/source/genwqe-user/lib'
Makefile:97: recipe for target 'lib' failed
make: *** [lib] Error 1

I tried to repro this and failed even before:

[haverkam@oc7383187364 genwqe-user]$ make CROSS=/usr/local/gcc-4.8.0-nolibc/powerpc64-linux/bin/powerpc64-linux-
make[1]: Entering directory `/home/haverkam/GenWQE/genwqe-git/genwqe-user/lib'
	[CC]	libzHW.o
In file included from libzHW.c:23:0:
/usr/local/gcc-4.8.0-nolibc/powerpc64-linux/lib/gcc/powerpc64-linux/4.8.0/include/stdint.h:9:26: fatal error: stdint.h: No such file or directory
 # include_next <stdint.h>
                          ^
compilation terminated.
make[1]: *** [libzHW.o] Fehler 1
make[1]: Leaving directory `/home/haverkam/GenWQE/genwqe-git/genwqe-user/lib'
make: *** [lib] Fehler 1

fhaverkamp, 2016-01-29:
We would need libcxl too.

vaibhav92, 2016-01-29:
I was able to cross compile genwqe over my self built cross tool chain using below command:

$ make CROSS=<cross-tool-prefix> PLATFORM=ppc64le

For me cross-tool-prefix = powerpc64le-unknown-linux-gnu- . This tool chain was built using crosstools-ng (http://crosstool-ng.org/).
Similar make command can be used for building libcxl.

~ Vaibhav

fhaverkamp, 2016-02-03:
How does libcxl behave in this case?

closed 2016-08-09 by fhaverkamp