ibm-genwqe/genwqe-user #76: Please rename csv2bin to a less generic name

xnox, 2016-02-16:
csv2bin binary name is too generic. It looks as it it's one of the <foo>2<bar> tools that converts generic one format to another (e.g. ps2ascii  ps2pdf  ps2pdf13  ps2pdfwr  ps2ps  ps2txt  ps2epsi  ps2pdf12 ps2pdf14  ps2pk  ps2ps2, etc.).

Is this actually a csv2vpd tool? Or it is specifically a genwqe-csv2vpd tool (only for genwqe cards)?

Please rename csv2bin to a more descriptive name.

xnox, 2016-02-16:
Reading sample spec file:
genwqe_csv2vpd

Would be a better name for this tool, similar to genwqe_vpdconf & genwqe_vpdupdate.

fhaverkamp, 2016-02-17:
Agreed, genwqe_csv2vpd is appropriate.

fhaverkamp, 2016-02-17:
After renaming the tool now twice ;-), I hopefully have now the correct name.

closed 2016-02-17 by fhaverkamp