ibm-genwqe/genwqe-user #77: Should set soname and dynamically link libzADC

xnox, 2016-02-16:
libzADC should be installed with a soname, into normal %{_libdir} location, with a soname.
Probably a matching soname, of the zlib library that is being mimicked.

genwqe_gzip
genwqe_gunzip
zlib_mt_perf

may not statically link libzADC.

fhaverkamp, 2016-02-26:
Gabriel, can you have a look how we can fix that, please?

fhaverkamp, 2016-04-15:
I found that we set the soname, but it points to the fully versioned name. I guess that is not right since it creates dependencies to previous versions and require relink once a new version comes out. I tried to fix that with: https://github.com/ibm-genwqe/genwqe-user/commit/9c7078a5acf1f3d883df045d446c58bfa3b7ad28

Resulting files are:
[haver@tul4eth3 genwqe-user]$ ll /usr/lib64/genwqe /usr/bin/genwqe* /usr/bin/zlib_mt_perf
-rwxr-xr-x. 1 root root  70464 15. Apr 14:49 /usr/bin/genwqe_cksum
-rwxr-xr-x. 1 root root 139176 15. Apr 14:49 /usr/bin/genwqe_echo
-rwxr-xr-x. 1 root root  70080 15. Apr 14:49 /usr/bin/genwqe_ffdc
-rwxr-xr-x. 1 root root 201392 15. Apr 14:49 /usr/bin/genwqe_gunzip
-rwxr-xr-x. 1 root root 201392 15. Apr 14:49 /usr/bin/genwqe_gzip
-rwxr-xr-x. 1 root root 138728 15. Apr 14:49 /usr/bin/genwqe_loadtree
-rwxr-xr-x. 1 root root 139008 15. Apr 14:49 /usr/bin/genwqe_maint
-rwxr-xr-x. 1 root root 139392 15. Apr 14:49 /usr/bin/genwqe_memcopy
-rwxr-xr-x. 1 root root   8362 15. Apr 14:49 /usr/bin/genwqe_mt_perf
-rwxr-xr-x. 1 root root 139112 15. Apr 14:49 /usr/bin/genwqe_peek
-rwxr-xr-x. 1 root root 139096 15. Apr 14:49 /usr/bin/genwqe_poke
-rwxr-xr-x. 1 root root  11608 15. Apr 14:49 /usr/bin/genwqe_test_gz
-rwxr-xr-x. 1 root root  70344 15. Apr 14:49 /usr/bin/genwqe_update
-rwxr-xr-x. 1 root root 206496 15. Apr 14:49 /usr/bin/zlib_mt_perf

/usr/lib64/genwqe:
insgesamt 204
lrwxrwxrwx. 1 root root     22 15. Apr 14:50 gunzip -> /usr/bin/genwqe_gunzip
lrwxrwxrwx. 1 root root     20 15. Apr 14:50 gzip -> /usr/bin/genwqe_gzip
-rwxr-xr-x. 1 root root 206216 15. Apr 14:49 libzADC-4.0.16.so
lrwxrwxrwx. 1 root root     17 15. Apr 14:50 libzADC.4.so -> libzADC-4.0.16.so
lrwxrwxrwx. 1 root root     17 15. Apr 14:50 libzADC.so -> libzADC-4.0.16.so
lrwxrwxrwx. 1 root root     17 15. Apr 14:50 libz.so -> libzADC-4.0.16.so
lrwxrwxrwx. 1 root root     17 15. Apr 14:50 libz.so.1 -> libzADC-4.0.16.so

I removed the lib duplication by fixing up the symlinks too. I hope that is better now.

fhaverkamp, 2017-05-30:
@xnox @krisman Can we close this?

xnox, 2017-06-14:
libzADC.4.so does not look right, should it not be libzADC.so.4 ?! and point to libzADC.so.4.0.16 ?

I haven't used this in a while, so I do need to pull latest master and retest this.

fhaverkamp, 2017-11-29:
lrwxrwxrwx 1 haver haver      26 Nov 29 12:05 libzADC.so -> libzADC.so.4.0.18-44-g3ff5
lrwxrwxrwx 1 haver haver      26 Nov 29 12:05 libzADC.so.4 -> libzADC.so.4.0.18-44-g3ff5
-rwxrwxr-x 1 haver haver  630230 Nov 29 12:05 libzADC.so.4.0.18-44-g3ff5

The git-id is not used for the distro builds. So I guess it is now as you mentioned libzADC.so.4 pointing to libzADC.so.4.0.18.