ibm-genwqe/genwqe-user #78: Does not Reproducibly Build, uses __DATE__ macro

xnox, 2016-02-16:
https://wiki.debian.org/ReproducibleBuilds/TimestampsFromCPPMacros

There is __DATE__ macro used in the code, which means it's impossible to reproduce the package build (ie. independently build the package again, and get identical binaries out with identical checksum hash).

Please consider removing __DATE__ macro, and instead e.g. use a full Version number, possibly with e.g. packaging version number that is with Debian Revision / RPM Revision&tag instead. That would still uniquely identify the build, but can be re-built without changing the output.

xnox, 2016-02-16:
For example, use GIT_VERSION instead of `__DATE__`

fhaverkamp, 2016-02-17:
https://github.com/ibm-genwqe/genwqe-user/commit/d31629f94f64b6ff746ac7ad71496ab350ff085e should fix that. I am using GIT_VERSION as you proposed.

closed 2016-02-17 by fhaverkamp