ibm-genwqe/genwqe-user #81: libzADC is underlinked, missing -ldl and -pthread

xnox, 2016-02-16:
libzADC is underlinked, missing -ldl and -pthread

fhaverkamp, 2016-02-18:
Hi xnox, did you find that by manual inspection, or did you use a tool to see that?

fhaverkamp, 2016-02-18:
[haverkam@oc7383187364 genwqe-user]$ ldd -r lib/libzADC.so 
	linux-vdso.so.1 =>  (0x00007ffdc3d6b000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f24b4180000)
	libdl.so.2 => /lib64/libdl.so.2 (0x00007f24b3f7c000)
	librt.so.1 => /lib64/librt.so.1 (0x00007f24b3d73000)
	libc.so.6 => /lib64/libc.so.6 (0x00007f24b39df000)
	/lib64/ld-linux-x86-64.so.2 (0x00000034ba400000)

I have added https://github.com/ibm-genwqe/genwqe-user/commit/7ee3f649c2cd95cf21e2d05561c6e9d96358a8b8 to solve this issue.

librt.so was missing too.

fhaverkamp, 2016-02-19:
I think with those changes it should be fixed. Reopen if this is not correct.

closed 2016-02-19 by fhaverkamp