ibm-genwqe/genwqe-user #102: workload mongod hangs when loading zEDC driver

hugulas, 2016-04-07:
Environment: SLES 12.1
Hardware: z13

We are using GENWQE built from GitHub, but the workload mongod hangs when loading it as follows. 

zBAOLX1:~# LD_LIBRARY_PATH=/usr/lib/genwqe/:$LD_LIBRARY_PATH ZLIB_DEFLATE_IMPL=1 ZLIB_TRACE=0x0f mongod --storageEngine wiredTiger --wiredTigerCollectionBlockCompressor zlib --dbpath /sys/fs/cgroup/zEDC_output/mongodb 
### _init: BUILD= ZLIB_TRACE=f ZLIB_INFLATE_IMPL=1 ZLIB_DEFLATE_IMPL=1 ZLIB_INFLATE_THRESHOLD=16384
sss Try loading software zlib "/opt/genwqe/lib/libz.so"
sss Try loading system software zlib "libz.so"
sss loading [zlibVersion]
sss Using dlopened libz.so
sss ZLIB_VERSION 1.2.8 (header version)

But the version 3.0.25 Frank sent me last summer works well with zEDC. Can we get latest stable GENWQE binary? Thanks!

Best Regards,
Qin Yue Chen
Mainframe Performance Analyst, VPA Developer

fhaverkamp, 2016-04-07:
Could we get an ltrace of the startup to see if we can figure out what breaks it? Which version of our code did you try (sha-id or tag)? Did you already try with our latest master branch?

fhaverkamp, 2016-04-07:
Hi hugulas,

to enable us using the software libz.so.1 installed in the distribution instead of statically linking in software libz, we use dlopen() to load the right symbols. Using LD_LIBRARY_PATH to point to /usr/lib/genwqe seems to cause now that our lib (/usr/lib/genwqe/libz.so) tries to load itself as software version, which does not work.

Please try to use LD_PRELOAD instead to activate our library. You do not need to set ZLIB_DEFLATE_IMPL=0x1 or alike anymore, since we assume that if you point to our hardware accelerated zlib, you of course intend to use the hardware if it is available.

zBAOLX1:~ # LD_PRELOAD=/usr/lib/genwqe/libz.so.1 ZLIB_TRACE=0x0f mongod --storageEngine wiredTiger --wiredTigerCollectionBlockCompressor zlib --dbpath /sys/fs/cgroup/zEDC_output/mongodb
### _init: BUILD= ZLIB_TRACE=f ZLIB_INFLATE_IMPL=1 ZLIB_DEFLATE_IMPL=1 ZLIB_INFLATE_THRESHOLD=16384
sss Try loading software zlib "/opt/genwqe/lib/libz.so"
sss Try loading system software zlib "libz.so"
sss loading [zlibVersion]
sss Using dlopened libz.so
sss   ZLIB_VERSION 1.2.8 (header version)
sss   zlibVersion  1.2.8 (libz.so version)
sss loading [deflateInit2_]
sss loading [deflateParams]
sss loading [deflateReset]
sss loading [deflatePrime]
sss loading [deflateCopy]
sss loading [deflate]
sss loading [deflateSetDictionary]
sss loading [deflateSetHeader]
sss loading [deflateEnd]
sss loading [inflateInit2_]
sss loading [inflateSync]
sss loading [inflatePrime]
sss loading [inflate]
sss loading [inflateReset]
sss loading [inflateReset2]
sss loading [inflateSetDictionary]
sss loading [inflateGetDictionary]
sss loading [inflateGetHeader]
sss loading [inflateEnd]
sss loading [inflateBackInit_]
sss loading [inflateBack]
sss loading [inflateBackEnd]
sss loading [gzopen]
sss loading [gzdopen]
sss loading [gzwrite]
sss loading [gzread]
sss loading [gzclose]
sss loading [gzflush]
sss loading [gzungetc]
sss loading [gzeof]
sss loading [gztell]
sss loading [gzerror]
sss loading [gzseek]
sss loading [gzputs]
sss loading [gzputc]
sss loading [gzgetc]
sss loading [gzputs]
sss loading [gzprintf]
sss loading [compress]
sss loading [uncompress]
sss loading [zError]
sss loading [zlibCompileFlags]
sss loading [adler32]
sss loading [adler32_combine]
sss loading [crc32]
sss loading [crc32_combine]
sss Software zlib 1.2.8/header 1.2.8 loaded
2016-04-07T15:11:48.760+0800 I CONTROL  [initandlisten] MongoDB starting : pid=27243 port=27017 dbpath=/sys/fs/cgroup/zEDC_output/mongodb 64-bit host=zBAOLX1
2016-04-07T15:11:48.760+0800 I CONTROL  [initandlisten] db version v3.2.0
2016-04-07T15:11:48.760+0800 I CONTROL  [initandlisten] git version: b80311d7fcfaae95c434ade551949921925234e6
2016-04-07T15:11:48.760+0800 I CONTROL  [initandlisten] allocator: system
2016-04-07T15:11:48.760+0800 I CONTROL  [initandlisten] modules: none
2016-04-07T15:11:48.760+0800 I CONTROL  [initandlisten] build environment:
2016-04-07T15:11:48.760+0800 I CONTROL  [initandlisten]     distarch: s390x
2016-04-07T15:11:48.760+0800 I CONTROL  [initandlisten]     target_arch: s390x
2016-04-07T15:11:48.760+0800 I CONTROL  [initandlisten] options: { storage: { dbPath: "/sys/fs/cgroup/zEDC_output/mongodb", engine: "wiredTiger", wiredTiger: { collectionConfig: { blockCompressor: "zlib" } } } }
2016-04-07T15:11:48.766+0800 I STORAGE  [initandlisten] exception in initAndListen: 29 Data directory /sys/fs/cgroup/zEDC_output/mongodb not found., terminating
2016-04-07T15:11:48.766+0800 I CONTROL  [initandlisten] dbexit:  rc: 100
zBAOLX1:~ # 

Well, I guess the data directory is missing, so you might need some more tweaking and change. Let me know if that works for you.

hugulas, 2016-04-08:
The solution works.
#** LD_PRELOAD**=/usr/lib/genwqe/libz.so.1 ZLIB_TRACE=0x0f mongod --storageEngine wiredTiger --wiredTigerCollectionBlockCompressor zlib --dbpath /sys/fs/cgroup/zEDC_output/mongodb

fhaverkamp, 2016-04-11:
Super. Thanks for reporting and trying it out. Let me know once you have some results. I am curious how well it worked out at the end.

closed 2016-04-11 by fhaverkamp