ibm-genwqe/genwqe-user #104: genwqe_test_gz: Ensure that path is correctly setup

fhaverkamp, 2016-04-14:
root@PC-Gzip:~/haver/genwqe-user# genwqe_test_gz -ACAPI 
genwqe_test_gz: ERROR: gzip and gunzip tools not found in /usr/bin/genwqe.
Make sure the GenWQELibZ RPM Package is installed and gzip and gunzip
are executable for root in /usr/bin/genwqe.

krisman, 2016-04-14:
Created symlinks in the deb package.  Is this specific to debian/ubuntu?  I don't see how this happen on rhel.  

Anyway, after fixing it, I'm getting another sort of error, 

│/usr/bin/genwqe_test_gz: line 193: 27893 Segmentation fault      ${GZIPTOOL} $file
│/usr/bin/genwqe_test_gz: line 193: 27890 Segmentation fault      ${GZIPTOOL} $file
│/usr/bin/genwqe_test_gz: line 193: 27888 Segmentation fault      ${GZIPTOOL} $file
│/usr/bin/genwqe_test_gz: line 193: 27900 Segmentation fault      ${GZIPTOOL} $file
│/usr/bin/genwqe_test_gz: line 193: 27924 Segmentation fault      ${GZIPTOOL} $file
│/usr/bin/genwqe_test_gz: line 193: 27932 Segmentation fault      ${GZIPTOOL} $file
│/usr/bin/genwqe_test_gz: line 193: 27916 Segmentation fault      ${GZIPTOOL} $file
│/usr/bin/genwqe_test_gz: line 193: 27902 Segmentation fault      ${GZIPTOOL} $file
│/usr/bin/genwqe_test_gz: line 193: 27889 Segmentation fault      ${GZIPTOOL} $file
│/usr/bin/genwqe_test_gz: line 193: 27930 Segmentation fault      ${GZIPTOOL} $file

Is this the same segfault you mentioned?

krisman, 2016-04-14:
With the symlinks this issue should be resolved. I'll leave it open so you see my question about the segfaults

closed 2016-04-15 by krisman