ibm-genwqe/genwqe-user #128: Negative value for Uncompressed ratio for genwqe_gzip test On Corsa PCI card

hathyaga, 2016-10-24:
===> Tested genwqe_gzip with a 108GB file:

[root@p8le03 home]# du -h test_file_harsha 
108G	test_file_harsha

[root@p8le03 home]# genwqe_gzip test_file_harsha 

[root@p8le03 home]# genwqe_gzip test_file_harsha.gz -l
         compressed        uncompressed  ratio uncompressed_name
         2583984192          -620756992  29.67% test_file_harsha

[root@p8le03 home]# du -h test_file_harsha.gz 
2.5G	test_file_harsha.gz

hathyaga, 2016-10-24:
@fhaverkamp  Issue for genwqe_gzip test for a negative value observed during compression test raised as requested.

fhaverkamp, 2016-10-25:
@hathyaga What was the resulting output of software gzip?
I wonder since the output size is stored at the end of the data. The field in rfc1952 is called ISIZE:

         ISIZE (Input SIZE)
            This contains the size of the original (uncompressed) input
            data modulo 2^32.

So I wonder if they print that or if they try to decompress, and discard the output to get the real value.

The spot in the code is this:
        if (!verbose) {
                fprintf(stderr,
                        "         compressed        uncompressed  ratio "
                        "uncompressed_name\n"
                        "%19lld %19d  %2.2f%% %s\n",
                        (long long)st.st_size, size, ratio,
                        out_f);
        } else {
                time_t t = time(NULL);
                struct tm *tm = localtime(&t);
                /* (const time_t *)&head.time */

                fprintf(stderr, "method  crc     date  time           "
                        "compressed        uncompressed  ratio "
                        "uncompressed_name\n"
                        "%s %x %s %2d %d:%d %19lld %19d  %2.2f%% %s\n",
                        "defla", crc32,
                        mon[tm->tm_mon], tm->tm_mday, tm->tm_hour, tm->tm_min,
                        (long long)st.st_size, size, ratio,
                        out_f);
        }
I guess some %d should better be %u or alike. We need to try.

fhaverkamp, 2016-10-25:
Here a temporary result:

haver@tul3eth3:~/genwqe-user.new$ ./tools/genwqe_gzip -ACAPI -B0 -l 3GiB.bin.gz 
         compressed        uncompressed  ratio uncompressed_name
         3223176858          3221225472  -0.06% 3GiB.bin
haver@tul3eth3:~/genwqe-user.new$ gzip -l 3GiB.bin.gz 
         compressed        uncompressed  ratio uncompressed_name
         3223176858          3221225472  -0.1% 3GiB.bin

closed 2016-11-07 by fhaverkamp