ibm-genwqe/genwqe-user #139: genwqe-user doesn't build on i686 platform

ngothan, 2017-02-13:
the 4.0.18 prerelease doesn't build on i686 platform. it breaks with the folling errors:

gcc -c  -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -m32 -march=i686 -fasynchronous-unwind-tables -DGIT_VERSION=\"4.0.18\" -I. -I../include -I../include/linux/uapi -D_GNU_SOURCE=1 -I../include -DCONFIG_DLOPEN_MECHANISM -DCONFIG_ZLIB_PATH=\"/usr/lib/libz.so\" -fPIC -fno-strict-aliasing software.c -o software.o
software.c:537:9: error: conflicting types for 'gzseek64'
 z_off_t gzseek64(gzFile file, z_off64_t offset, int whence)
         ^~~~~~~~
In file included from software.c:23:0:
/usr/include/zlib.h:1840:30: note: previous declaration of 'gzseek64' was here
    ZEXTERN z_off64_t ZEXPORT gzseek64 OF((gzFile, z_off64_t, int));
                              ^~~~~~~~
software.c:553:9: error: conflicting types for 'gzoffset64'
 z_off_t gzoffset64(gzFile file)
         ^~~~~~~~~~
In file included from software.c:23:0:
/usr/include/zlib.h:1842:30: note: previous declaration of 'gzoffset64' was here
    ZEXTERN z_off64_t ZEXPORT gzoffset64 OF((gzFile));
                              ^~~~~~~~~~
make[1]: *** [Makefile:209: software.o] Error 1

fhaverkamp, 2017-03-13:
Can you please have a look if Brenos changes here:
https://github.com/ibm-genwqe/genwqe-user/commit/3c768b385a6a256505a1a569fed0c73f9aa82a9b
are fixing the problem you are reporting? Please close this if it is a good fix. Thanks.

ngothan, 2017-03-13:
i confirm the fix. Now it builds fine on 32-bit arches. Thanks

closed 2017-03-13 by ngothan