ibm-genwqe/genwqe-user #148: Ubuntu 16.04: LD_PRELOAD needed when using genwqe_gzip/gunzip

fhaverkamp, 2017-05-30:
haver@p95a17:~$ apt-cache search genwqe
genwqe-libz-dev - Development tools and libraries for IBM GenWQE PCI gzip and IBM CAPI.
genwqe-libz - Libraries to use IBM GenWQE card and CAPI gzip
genwqe-tools - utilities for accelerated libz implementation
haver@p95a17:~$ genwqe_gzip
genwqe_gzip: error while loading shared libraries: libzADC.so.4: cannot open shared object file: No such file or directory
haver@p95a17:~$ LD_PRELOAD=/usr/lib/powerpc64le-linux-gnu/genwqe/libzADC.so genwqe_gzip -h
Usage: genwqe_gzip [OPTION]... [FILE]...
Compress or uncompress FILEs (by default, compress FILES in-place).

Mandatory arguments to long options are mandatory for short options too.

  -c, --stdout      write on standard output, keep original files unchanged
  -d, --decompress  decompress
  -f, --force       force overwrite of output file and compress links
  -h, --help        give this help
  -l, --list        list compressed file contents
  -L, --license     display software license
  -N, --name        save or restore the original name and time stamp
  -q, --quiet       suppress all warnings
  -S, --suffix=SUF  use suffix SUF on compressed files
  -v, --verbose     verbose mode
  -V, --version     display version number
  -1, --fast        compress faster
  -9, --best        compress better

Special options for testing and debugging:
  -A, --accelerator-type=GENWQE|CAPI CAPI is only available for System p
  -B, --card=<card_no> -1 is for automatic card selection
  -X, --cpu <cpu>   force to run on CPU <cpu>
  -s, --software    force to use software compression/decompression
  -i, --i_bufsize   input buffer size (128 KiB)
  -o, --o_bufsize   output buffer size (128 KiB)
  -N, --name=NAME   write NAME into gzip header
  -C, --commente=CM write CM into gzip header
  -E, --extra=EXTRA write EXTRA (file) into gzip header

With no FILE, or when FILE is -, read standard input.

NOTE: Not all options are supported in this limited version!
Suggestions or patches are welcome!

Report bugs via https://github.com/ibm-genwqe/genwqe-user.

Called with:
  ARGV[0]: "genwqe_gzip"
  ARGV[1]: "-h"

We might need to move libzADC.so.1 into /usr/lib to get this right and follow the proposal in the other bug, where this was already proposed.

fhaverkamp, 2017-11-29:
@guilhermepiccoli May you have a look please?

maurorodrigues, 2018-04-18:
Hi @fhaverkamp I'll start to look on this. For some reason though I can't change the assignee to myself.