ibm-genwqe/genwqe-user #154: zlib inflate error on git log and git status (HW version)

mdebatin, 2017-08-14:
Hi,

I have seen a zlib inflate error for the following scenario:

**Problem:** zlib inflate error 

**Scenario to recreate:**
SLES 12.2 ppc64le
Git version: 2.6.3
export parameters:
export ZLIB_ACCELERATOR=GENWQE 
export ZLIB_TRACE=0xf
export ZLIB_VERBOSE=0
export ZLIB_DEFLATE_IMPL=1
export ZLIB_INFLATE_IMPL=1
export LD_PRELOAD=#dynamically linked library compiled from the following git commit:
03ca0f1372277bdba4dae699e4b3fe5ca0ed84e1

execution command:
execute git log or git status on any git repository, e.g. use the current genwqe-user git repo.

**traces:**
git log:
[zlib_trace_git_log.txt](https://github.com/ibm-genwqe/genwqe-user/files/1221610/zlib_trace_git_log.txt)

git status:
[zlib_trace_git_status.txt](https://github.com/ibm-genwqe/genwqe-user/files/1221611/zlib_trace_git_status.txt)

**Note:**
Software version:
export ZLIB_DEFLATE_IMPL=0
export ZLIB_INFLATE_IMPL=0

runs fine

Kind regards,

Maurice

guilhermepiccoli, 2017-10-06:
Maurice, thanks for your report! Frank and I are investigating this...we could narrow the issue, seems that a user-memory to DMA allocation is failing. I'll need to analyze it deeply to understand (I'm not too agile in this investigation since I have tons of POWER9 things to investigate too, so sorry for any delay).

fhaverkamp, 2017-10-16:
Meanwhile we found a problem when mapping read only pages in our genwqe-card Linux driver. Guilherme worked out a fix  for that, which we are not starting to test.

mdebatin, 2017-10-16:
thanks - sounds good - I guess you ment to write "which we are noW starting to test". Unfortunately I cannot test right now since we have to manage our infrastructure. I don't have access to the FPGA cards right now.

fhaverkamp, 2017-10-27:
https://lkml.org/lkml/2017/10/20/771

fhaverkamp, 2017-11-29:
@guilhermepiccoli Is this meanwhile in the distro kernel or do you need to trigger an update?

fhaverkamp, 2017-12-01:
Fixed via kernel driver patch mentioned above. Test-case is in place now.

closed 2017-12-01 by fhaverkamp

guilhermepiccoli, 2017-12-14:
> @guilhermepiccoli Is this meanwhile in the distro kernel or do you need to trigger an update?

Frank, I'll check SLES kernel and open a bugzilla accordingly. I'll request it fro SLES 12 and 15, ok?

Thanks,

Guilherme

fhaverkamp, 2018-01-08:
> thread [3].
> Links:

>> @guilhermepiccoli [1] Is this meanwhile in the distro kernel or do
>> you need to trigger an update?
> 
> Frank, I'll check SLES kernel and open a bugzilla accordingly. I'll
> request it fro SLES 12 and 15, ok?

Fine with me. Thanks.