ibm-genwqe/genwqe-user #156: Potential arbitrary code execution bug found by Coverity scan

kholia, 2017-11-15:
I ran Coverity against the genwqe-tools package in RHEL 74. This corresponds to upstream "v4.0.18.tar.gz".

In particular the finding below is a bit troubling,

```
Error: TAINTED_STRING (CWE-20):
genwqe-user-4.0.18/lib/software.c:597: tainted_string_return_content: "getenv" returns tainted string content.
genwqe-user-4.0.18/lib/software.c:597: var_assign: Assigning: "zlib_path" = "getenv("ZLIB_PATH")", which taints "zlib_path".
genwqe-user-4.0.18/lib/software.c:603: tainted_string: Passing tainted string "zlib_path" to "dlopen", which cannot accept tainted data.
#  601|   		sw_trace("Loading software zlib \"%s\"\n", zlib_path);
#  602|   		dlerror();
#  603|-> 		handle = dlopen(zlib_path, RTLD_LAZY);
#  604|   		if (handle != NULL)
#  605|   			goto load_syms;
```

Coverity is saying that this particular code can allow arbitrary code execution in certain circumstances.

The code in the affected function ("zedc_sw_init") is trying to find the right libz.so.1 on the system. 

According to a comment before this function, 

```
/**
 * RHEL7.2:
 *   $ ldconfig -p | grep libz.so.1 | cut -d' ' -f4 | head -n1
 *   /lib64/libz.so.1
 * ...
 * Intel with RHEL6.7:
 *   $ ldconfig -p | grep libz.so.1 | cut -d' ' -f4 | head -n1
 *   /lib64/libz.so.1
```

... the path to `libz.so.1` seems to be fixed on various RHEL versions. So would it be possible for downstream consumers (e.g. RHEL) of this package to simplify this code, and remove the `zlib_path" = "getenv("ZLIB_PATH")` functionality safely? Can this removal be done in the upstream code by you?

If there is any possibility (now and in the future) that the `zedc_sw_init` code could be run with system level privileges, we need to harden this particular code now.

I have been informed that  `genwqe_peek` and `genwqe_poke` tools need to be run as root because both tools reads and writes specific registers.

fhaverkamp, 2017-11-24:
Hi, thanks for making us aware of this problematic spot in the code. Originally we used it for debugging and to enable us switching between different zlib software installations. I will discuss internally what we do with it. Maybe ifdefing it and just keep it for debugging purposes for special builds.

fhaverkamp, 2017-11-29:
I have put the following patch in place to address the issue:
    https://github.com/ibm-genwqe/genwqe-user/commit/f4a0a7c39ea94c3a2abef5019f5b41e2628efa31

We need to test this now, before it can be included. Please feel free to review and drop a comment if you think that more change is required. Thanks.

kholia, 2017-11-29:
Thanks Frank! The fix looks good to me. I will request our internal package maintainer to double-check this.

ngothan, 2017-11-29:
@fhaverkamp , it also looks fine to me. thanks