ibm-genwqe/genwqe-user #157: inflate() returns Z_OK instead of Z_STREAM_END in TokuDB

davidzengxhsh, 2017-11-26:
During the integrating CAPI GZIP lib with TokuDB, we found that inflate() Z_OK instead of Z_STREAM_END. And the unexpected return value cause a assertion in TokuDB.

In order to duplicate this problem, we dumped the data and write a test code as attached.

toku_test.bin - This is the compressed data dumped from TokuDB.
zpipe_example.c - This is a modified decompress code to duplicate the issue.

Compile the code: gcc zpipe_example.c -o zpipe_example -L /opt/at10.0/lib64/power8/:/usr/lib64/ -lz
Run with software zlib; ./zpipe_example -d --format=DEFLATE -o 139264 -f tmp_sw.txt
		inflate return:1(expected 1)
		bytes write to output file:139264

Run with hardware zlib: ZLIB_ACCELERATOR=CAPI LD_PRELOAD=/usr/lib/genwqe/libz.so.1 ./zpipe_example -d --format=DEFLATE -o 139264 -f tmp_hw.txt
		00004172.00004172 inflate.c:1227: Warn: [zedc_inflate] What a pity, we guessed wrong and need to repeat
		inflate return:0(expected 1)
		bytes write to output file:139264

[test.zip](https://github.com/ibm-genwqe/genwqe-user/files/1503574/test.zip)

fhaverkamp, 2017-11-29:
Thanks for reporting the issue. I have put some fixes in place in the following branch:
    https://github.com/ibm-genwqe/genwqe-user/tree/fix_for_deflateBound

Please have a look of those work for you. After a fix on the fix our internal test-case seems to work well too.

fhaverkamp, 2017-12-01:
Should be fine now. Can we close this?

davidzengxhsh, 2017-12-10:
Thanks, Frank! It's fine now with your fix.

closed 2017-12-10 by davidzengxhsh