ibm-genwqe/genwqe-user #159: Software fallback to SAP libz fails with compression error

mdebatin, 2017-12-01:
Scenario:
SAP Hana 2.0, libzADC dynamically linked to Hana, dynamical fallback to the SAP libz library.

Bug:
Compression errors

Detailed description will follow. I will try to produce a "minimal not working example".

fhaverkamp, 2017-12-05:
Hi Maurice, do you have any more debug hints, traces or even example code yet?

mdebatin, 2017-12-06:
No, unfortunately I don't have the time to address this issue up right now

fhaverkamp, 2018-03-08:
@mdebatin Is this still needed? Or should we rather close the issue and reopen/create new if it reoccurs and if we have a procedure to reproduce this?

mdebatin, 2018-03-08:
Hi, I think for now we can close the issue and reopen it if it occurs in another scenario.

closed 2018-11-26 by fhaverkamp